osrf/ros2_test_cases #1006: Testing Main

Yadunund, 2023-05-01:
Check the documentation for the 'Testing Main' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Testing Main page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Testing/Testing-Main.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-06:
On a debian jammy installation with fastDDS.

I have checked the 4 documentation pages regarding Testing (Main, Running, Writing in Python, Writing in C++).
I have been able to follow the documentation, it was clear to me, and I have not seen any obvious error.
I have been able to create/compile and run tests in both Python and C++.

clalancette, 2023-05-06:
Thanks for testing!

closed 2023-05-06 by clalancette