osrf/ros2_test_cases #1008: The Keystore

Yadunund, 2023-05-01:
Check the documentation for the 'The Keystore' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [The Keystore page](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/The-Keystore.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-09:
Assigned to myself.

I can confirm the following:

[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

Yadunund, 2023-05-10:
Great, thanks for testing!

closed 2023-05-10 by Yadunund