osrf/ros2_test_cases #1009: Time Travel With Tf2 Cpp

Yadunund, 2023-05-01:
Check the documentation for the 'Time Travel With Tf2 Cpp' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Time Travel With Tf2 Cpp page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastDDS.

I have followed the documentation for the "Time Travel With Tf2 Cpp", it was clear to me but have small inaccuracies in relation to modifying the code of `turtle_tf2_listener.cpp`

This page is more accurate than the "using Time C++", as the modified code properly use the variable name `t`.
So updates is perhaps unnecessary, we could understand the requested code modification as modification to the content within the `try` block.

If we want to put the time variable computations outside the `try` block, here are the updated code references:

```
        rclcpp::Time when = this->get_clock()->now() - rclcpp::Duration(5, 0);
        try {
          t = tf_buffer_->lookupTransform(
            toFrameRel, fromFrameRel,
            when,
            50ms);
        } catch (const tf2::TransformException & ex) {
            ...
```

```
        rclcpp::Time now = this->get_clock()->now();
        rclcpp::Time when = now - rclcpp::Duration(5, 0);
        try {
          t = tf_buffer_->lookupTransform(
            toFrameRel,
            now,
            fromFrameRel,
            when,
            "world",
            50ms);
        } catch (const tf2::TransformException & ex) {
            ...
```

Yadunund, 2023-05-11:
Thanks for trying this out. I left a comment in your previous PR to also update this section of the tutorial with your findings.

closed 2023-05-11 by Yadunund

BorisBoutillier, 2023-05-11:
Good idea. I have added the fixes from this testcase to the PR: https://github.com/ros2/ros2_documentation/pull/3583