osrf/ros2_test_cases #1010: Time Travel With Tf2 Py

Yadunund, 2023-05-01:
Check the documentation for the 'Time Travel With Tf2 Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Time Travel With Tf2 Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Time-Travel-With-Tf2-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installtion with fastDDS.

I have been able to follow the documentation for "Time Travel with Tf3 Py" page, it was cleared to me, but has some inaccuracies in regards to the code of `turtle_tf2_listener.py` it explains to modify.

I would update the two code sections that ask to edit `turtle_tf2_listener.py` with:
```
                when = self.get_clock().now() - rclpy.time.Duration(seconds=5.0)
                try:
                    t = self.tf_buffer.lookup_transform(
                        to_frame_rel,
                        from_frame_rel,
                        when,
                        timeout=rclpy.duration.Duration(seconds=0.05))
                except TransformException as ex:
                        ...
```
and
```
                when = self.get_clock().now() - rclpy.time.Duration(seconds=5.0)
                try:
                    t = self.tf_buffer.lookup_transform_full(
                        target_frame=to_frame_rel,
                        target_time=rclpy.time.Time(),
                        source_frame=from_frame_rel,
                        source_time=when,
                        fixed_frame='world',
                        timeout=rclpy.duration.Duration(seconds=0.05))
                except TransformException as ex:
                        ...
```

Yadunund, 2023-05-10:
Thanks for the feedback! Do you mind opening a PR to ros2/ros2_documentation with the proposed changes?

BorisBoutillier, 2023-05-10:
I'll try to do that !

clalancette, 2023-05-11:
Given that we have the PR open, I'm going to go ahead and close this one.

Thank you for testing!

closed 2023-05-11 by clalancette