osrf/ros2_test_cases #1011: Topic Statistics Tutorial

Yadunund, 2023-05-01:
Check the documentation for the 'Topic Statistics Tutorial' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Topic Statistics Tutorial page](https://docs.ros.org/en/iron/Tutorials/Advanced/Topic-Statistics-Tutorial/Topic-Statistics-Tutorial.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-06:
Assigned to myself.

Can confirm all 3 checks:

[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

[image]

clalancette, 2023-05-06:
Thank you for testing!

closed 2023-05-06 by clalancette