osrf/ros2_test_cases #1013: Understanding ROS2 Nodes

Yadunund, 2023-05-01:
Check the documentation for the 'Understanding ROS2 Nodes' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Understanding ROS2 Nodes page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Nodes/Understanding-ROS2-Nodes.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
I'll take this

clalancette, 2023-05-03:
Thanks for testing!

closed 2023-05-03 by clalancette