osrf/ros2_test_cases #1014: Understanding ROS2 Parameters

Yadunund, 2023-05-01:
Check the documentation for the 'Understanding ROS2 Parameters' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Understanding ROS2 Parameters page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Parameters/Understanding-ROS2-Parameters.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
I'll take this

tylerjw, 2023-05-03:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

clalancette, 2023-05-03:
> This seems weird to me. Boolean parameters can have integer constraints?

Hm.  This is kind of a bug in two parts.  First of all, the code in https://github.com/ros/ros_tutorials/blob/cac3e244da4a3df7cefebd88f1469750249b4e8c/turtlesim/src/turtle_frame.cpp#L82 should not set a range at all; that is a boolean parameter as you point out.  This PR fixes that: https://github.com/ros/ros_tutorials/pull/150

However, the larger question is whether we should have these kind of checks in rclcpp so you aren't even allowed to get into this situation.  I'd think that would be a good feature request, though I fear we are going to have to do this checking at runtime, rather than compile-time.  In any case, @tylerjw if you could open a feature request on https://github.com/ros2/rclcpp , that would be much appreciated.

clalancette, 2023-05-04:
We've merged the fix for `turtlesim` into both rolling and iron.  And we opened up the follow-up bug with rclcpp. So I'm going to close this one out.  Thanks for testing!

closed 2023-05-04 by clalancette