osrf/ros2_test_cases #1016: Understanding ROS2 Topics

Yadunund, 2023-05-01:
Check the documentation for the 'Understanding ROS2 Topics' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Understanding ROS2 Topics page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
I'll take this

tylerjw, 2023-05-03:
I don't think this is correct (It differs from the image in the tutorial):
[image]

tylerjw, 2023-05-03:
The tutorial makes sense considering the notes I left above. Here is the output when sending ctl-C to close rqt_graph:

```
rclpy.shutdown() was called before QThread.quit()
PluginHandler.shutdown_plugin() plugin "rqt_graph/RosGraph#0" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 136, in shutdown_plugin
    self._shutdown_plugin()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler_direct.py", line 92, in _shutdown_plugin
    self.emit_shutdown_plugin_completed()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 150, in emit_shutdown_plugin_completed
    callback(self._instance_id)
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 480, in _close_application_shutdown_callback
    self._close_application_signal()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 483, in _close_application_signal
    self._plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 62, in shutdown
    self._destroy_node()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 89, in _destroy_node
    rclpy.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:313

^C[INFO] [1683135438.923384012] [rclcpp]: signal_handler(signum=2)
Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 127 in ./src/class_loader.cpp
```

Yadunund, 2023-05-04:
> I don't know what causes this but the behavior of the `Hide` checkboxes is not consistent.
> 
> * Some of them cause a refresh to happen when you toggle them: `Debug` and `Dead sinks`
> * Some of them do not refresh when you toggle them: `Params`

Thanks for identifying this problem. I've opened https://github.com/ros-visualization/rqt_graph/pull/86 which should refresh when toggling `Params`

Yadunund, 2023-05-04:
Also tracking other issue reported https://github.com/osrf/ros2_test_cases/discussions/1048

telemething, 2023-05-09:
I'd like to take a look too.

Yadunund, 2023-05-18:
@tylerjw I tried several times but I couldn't reproduce the weird circles. @telemething did you encounter the same?

Anyways, we're logged the circle issue in the discussions board so I'll close this ticket for now.

closed 2023-05-18 by Yadunund