osrf/ros2_test_cases #1017: URDF Main

Yadunund, 2023-05-01:
Check the documentation for the 'URDF Main' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [URDF Main page](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/URDF-Main.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-08:
This is just an index page with links to the URDF tutorials.
All links seem ok.

clalancette, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by clalancette