osrf/ros2_test_cases #1018: Using Parameters In A Class CPP

Yadunund, 2023-05-01:
Check the documentation for the 'Using Parameters In A Class CPP' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Using Parameters In A Class CPP page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-CPP.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Taking this.

marioprats, 2023-05-04:
Works as expected.
As with the python one, I created PR https://github.com/ros2/ros2_documentation/pull/3529 to clarify the expected output.

Yadunund, 2023-05-05:
Awesome, thanks for testing!

closed 2023-05-05 by Yadunund