osrf/ros2_test_cases #1019: Using Parameters In A Class Python

Yadunund, 2023-05-01:
Check the documentation for the 'Using Parameters In A Class Python' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Using Parameters In A Class Python page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-Python.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Tested this and works as expected.
Sent PR https://github.com/ros2/ros2_documentation/commit/f8be30e9b12cc54ce32e32a9efaf58b5e217570d to clarify the expected output in the tutorial.

Yadunund, 2023-05-05:
Great thanks for testing!

closed 2023-05-05 by Yadunund