osrf/ros2_test_cases #1020: Using Python Packages

Yadunund, 2023-05-01:
Check the documentation for the 'Using Python Packages' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Using Python Packages page](https://docs.ros.org/en/iron/Tutorials/Using-Python-Packages.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

avanmalleghem, 2023-05-09:
I'm on it

avanmalleghem, 2023-05-09:
> Check the documentation for the 'Using Python Packages' page
> 
> ## Setup
> * DDS vendor: FastDDS
> * BuildType: Debian
> * Os: Ubuntu Jammy
> * Chip: Amd64
> 
> ## Links
> * [Using Python Packages page](https://docs.ros.org/en/iron/Tutorials/Using-Python-Packages.html)
> 
> ## Checks
> * [x]  **I was able to follow the documentation.**
> * [ ]  **The documentation seemed clear to me.**
> * [x]  **The documentation didn't have any obvious errors.**
> 
> _You can find the code used to generate this test case [here](https://github.com/audrow/yatm)_

Everything seems good but the last note saying :
"If you want release your package on Bloom, you should to add the packages you require to rosdep, see the [rosdep key contribution guide](http://docs.ros.org/en/independent/api/rosdep/html/contributing_rules.html)."

Solution :
"If you want **to** release your package on Bloom, you should ~~to~~ add the packages you require to rosdep, see the [rosdep key contribution guide](http://docs.ros.org/en/independent/api/rosdep/html/contributing_rules.html)."

closed 2023-05-09 by Yadunund