osrf/ros2_test_cases #1021: Using Rqt Console

Yadunund, 2023-05-01:
Check the documentation for the 'Using Rqt Console' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Using Rqt Console page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Using-Rqt-Console/Using-Rqt-Console.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

tylerjw, 2023-05-03:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

Followed the tutorial and it all worked as expected.

tylerjw, 2023-05-03:
When shutting down rqt_console with Ctl-C it does this:

```
tyler@viable-velociraptor:~/code/ws_iron_overlay$ ros2 run rqt_console rqt_console
^C[INFO] [1683134102.495117773] [rclcpp]: signal_handler(signum=2)
rclpy.shutdown() was called before QThread.quit()
PluginHandler.shutdown_plugin() plugin "rqt_console/Console#0" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 136, in shutdown_plugin
    self._shutdown_plugin()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler_direct.py", line 92, in _shutdown_plugin
    self.emit_shutdown_plugin_completed()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 150, in emit_shutdown_plugin_completed
    callback(self._instance_id)
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 480, in _close_application_shutdown_callback
    self._close_application_signal()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 483, in _close_application_signal
    self._plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 62, in shutdown
    self._destroy_node()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 89, in _destroy_node
    rclpy.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:313
```

Yadunund, 2023-05-04:
Thanks for testing this out. I've documented the rqt issue here https://github.com/osrf/ros2_test_cases/discussions/1048

closed 2023-05-04 by Yadunund