osrf/ros2_test_cases #1022: Using Stamped Datatypes With Tf2 Ros MessageFilter

Yadunund, 2023-05-01:
Check the documentation for the 'Using Stamped Datatypes With Tf2 Ros MessageFilter' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Using Stamped Datatypes With Tf2 Ros MessageFilter page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Using-Stamped-Datatypes-With-Tf2-Ros-MessageFilter.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-09:
On a debian jammy installation with fastDDS.

I have done the "Using Stamped Datatypes with Tf2 Ros MessageFilter" tutorial.
I was able to follow the documentation, it was clear to me , and I did not see any error in wording/code or hyperlinks.

[image]

Yadunund, 2023-05-09:
Thanks for testing!

closed 2023-05-09 by Yadunund