osrf/ros2_test_cases #1024: Using Xacro to Clean Up a URDF File

Yadunund, 2023-05-01:
Check the documentation for the 'Using Xacro to Clean Up a URDF File' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Using Xacro to Clean Up a URDF File page](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-Xacro-to-Clean-Up-a-URDF-File.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Taking this.

marioprats, 2023-05-04:
All good.

clalancette, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by clalancette