osrf/ros2_test_cases #1025: Webots

Yadunund, 2023-05-01:
Check the documentation for the 'Webots' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Webots page](https://docs.ros.org/en/iron/Tutorials/Advanced/Simulators/Webots.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

omichel, 2023-05-04:
We would love to proceed with this one, based on the new version of the Webots ROS 2 documentation which is currently under review since March 21st, 2023 at https://github.com/ros2/ros2_documentation/pull/3357
I would truly appreciate if this PR could get reviewed and hopefully merged, so that we can proceed with the tests on the new official version of the tutorial. Testing a version of the documentation which is not yet accepted and subject to changes seems counterproductive to me...

ygoumaz, 2023-05-23:
Hi, I have tested the complete tutorial on the Iron distribution and it works perfectly. The examples in our package also work with very minimal tweaking. This issue can be closed as complete.

clalancette, 2023-05-23:
Thank you for testing!