osrf/ros2_test_cases #1026: Working with multiple RMW implementations

Yadunund, 2023-05-01:
Check the documentation for the 'Working with multiple RMW implementations' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Working with multiple RMW implementations page](https://docs.ros.org/en/iron/Tutorials/Working-with-multiple-RMW-implementations.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

iuhilnehc-ynos, 2023-05-09:
* [x]  **I was able to follow the documentation.**
* [x]  **The documentation seemed clear to me.**
* [ ]  **The documentation didn't have any obvious errors.**

I am not able to rebuild the rmw_connextcpp by using `--cmake-force-configure` if it's built before.

<details><summary> built with `--cmake-force-configure`, but not to generate librmw_connextdds.so </summary>
<p>

```shell
[10:33:01:043264397 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2_test_cases-1026 
chenlh ros2_test_cases-1026 $ . /opt/ros/iron/setup.bash 
[10:33:06:686088373 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2_test_cases-1026 
chenlh ros2_test_cases-1026 $ colcon build
Starting >>> rti_connext_dds_cmake_module
--- stderr: rti_connext_dds_cmake_module                         
CMake Warning at cmake/rti_build_helper.cmake:380 (message):
  no CONNEXTDDS_DIR nor NDDSHOME specified
Call Stack (most recent call first):
  cmake/rti_build_helper.cmake:554 (rti_load_connextddsdir)
  CMakeLists.txt:25 (rti_find_connextpro)

---
Finished <<< rti_connext_dds_cmake_module [0.66s]
Starting >>> rmw_connextdds_common
--- stderr: rmw_connextdds_common                         
CMake Warning at /home/chenlh/Projects/ROS2/ros2_test_cases-1026/install/rti_connext_dds_cmake_module/share/rti_connext_dds_cmake_module/cmake/rti_build_helper.cmake:380 (message):
  no CONNEXTDDS_DIR nor NDDSHOME specified
Call Stack (most recent call first):
  /home/chenlh/Projects/ROS2/ros2_test_cases-1026/install/rti_connext_dds_cmake_module/share/rti_connext_dds_cmake_module/cmake/rti_build_helper.cmake:554 (rti_load_connextddsdir)
  CMakeLists.txt:185 (rti_find_connextpro)

---
Finished <<< rmw_connextdds_common [0.72s]
Starting >>> rmw_connextdds
Starting >>> rmw_connextddsmicro
Finished <<< rmw_connextdds [0.68s]                                                               
Finished <<< rmw_connextddsmicro [0.69s]

Summary: 4 packages finished [2.23s]
  2 packages had stderr output: rmw_connextdds_common rti_connext_dds_cmake_module
[10:33:11:611307149 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2_test_cases-1026 
chenlh ros2_test_cases-1026 $ source /opt/rti.com/rti_connext_dds-6.0.1/resource/scripts/rtisetenv_x64Linux4gcc7.3.0.bash
###############################################################################

         (c) Copyright, Real-Time Innovations, All rights reserved.            
                                                                               
                           RTI Connext DDS 6.0.1                       

###############################################################################
[10:33:36:694798765 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2_test_cases-1026 
chenlh ros2_test_cases-1026 $ colcon build --cmake-force-configure
Starting >>> rti_connext_dds_cmake_module
Finished <<< rti_connext_dds_cmake_module [0.25s]                
Starting >>> rmw_connextdds_common
Finished <<< rmw_connextdds_common [0.30s]                
Starting >>> rmw_connextdds
Starting >>> rmw_connextddsmicro
Finished <<< rmw_connextdds [0.31s]                                                               
Finished <<< rmw_connextddsmicro [0.31s]

Summary: 4 packages finished [1.02s]
[10:33:47:387027197 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2_test_cases-1026 
chenlh ros2_test_cases-1026 $ ls build/rmw_connextdds
ament_cmake_core               ament_cmake_uninstall_target  CMakeFiles                      colcon_command_prefix_build.sh.env  install_manifest.txt
ament_cmake_environment_hooks  AMENT_IGNORE                  cmake_install.cmake             CTestConfiguration.ini              Makefile
ament_cmake_index              cmake_args.last               colcon_build.rc                 CTestCustom.cmake
ament_cmake_package_templates  CMakeCache.txt                colcon_command_prefix_build.sh  CTestTestfile.cmake
[10:33:57:212011580 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2_test_cases-1026 
chenlh ros2_test_cases-1026 $
[... 30 lines cut ...]
ament_cmake_export_libraries     ament_cpplint                 CMakeCache.txt       CTestConfiguration.ini
ament_cmake_index                AMENT_IGNORE                  CMakeFiles           CTestCustom.cmake
ament_cmake_package_templates    ament_lint_cmake              cmake_install.cmake  CTestTestfile.cmake

```

</p>
</details>

There is no option named `--packages-start` in colcon.
<details><summary> error log </summary>
<p>

```shell

[10:51:20:347109028 OptiPlex-7080] /home/chenlh/Projects/ROS2/ros2-iron 
chenlh ros2-iron $ RMW_IMPLEMENTATION=rmw_connextdds colcon build --cmake-clean-cache --cmake-force-configure --start-with rmw
usage: colcon [-h] [--log-base LOG_BASE] [--log-level LOG_LEVEL]
              {build,extension-points,extensions,graph,info,list,metadata,mixin,test,test-result,version-check} ...
colcon: error: unrecognized arguments: --start-with rmw

```

</p>
</details>

Yadunund, 2023-05-11:
I see that the documentation has been fixed, thanks for that. Will close this ticket.

closed 2023-05-11 by Yadunund