osrf/ros2_test_cases #1027: Writing a Cpp Action Server Client

Yadunund, 2023-05-01:
Check the documentation for the 'Writing a Cpp Action Server Client' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing a Cpp Action Server Client page](https://docs.ros.org/en/iron/Tutorials/Actions/Writing-a-Cpp-Action-Server-Client.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Taking the action server tutorials.

marioprats, 2023-05-04:
Everything working as expected.

clalancette, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by clalancette