osrf/ros2_test_cases #1030: Writing A Simple Cpp Service And Client

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Simple Cpp Service And Client' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Simple Cpp Service And Client page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Cpp-Service-And-Client.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jjd9, 2023-05-04:
May I take this one?

jjd9, 2023-05-04:
The instructions were very clear and seemed free of errors. I copied the code, as instructed, compiled and ran it, and the service and client worked as expected.
```
johnd@johnd-X510UQ:~/Desktop/ros2_ws$ ros2 run cpp_srvcli server
[INFO] [1683161214.502770082] [rclcpp]: Ready to add two ints.
[INFO] [1683161232.415689917] [rclcpp]: Incoming request
a: 2 b: 3
[INFO] [1683161232.415787593] [rclcpp]: sending back response: [5]
```

```
johnd@johnd-X510UQ:~/Desktop/ros2_ws$ ros2 run cpp_srvcli client 2 3
[INFO] [1683161232.416465108] [rclcpp]: Sum: 5
```

Yadunund, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by Yadunund