osrf/ros2_test_cases #1031: Writing A Simple Py Publisher And Subscriber

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Simple Py Publisher And Subscriber' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Simple Py Publisher And Subscriber page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

SrijaneeBiswas, 2023-05-04:
Assigned to myself and can successfully confirm the mentioned checks:

[x]: I was able to follow the documentation.
[x]: The documentation seemed clear to me.
[x]: The documentation didn't have any obvious errors.

Here's a screenshot of the tutorial running on my terminal:
[image]

clalancette, 2023-05-04:
Thank you for testing!