osrf/ros2_test_cases #1032: Writing A Simple Py Service And Client

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Simple Py Service And Client' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Simple Py Service And Client page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Service-And-Client.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Everything worked as expected.

clalancette, 2023-05-04:
Thanks for testing!