osrf/ros2_test_cases #1033: Writing A Tf2 Broadcaster Cpp

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Tf2 Broadcaster Cpp' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Tf2 Broadcaster Cpp page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Broadcaster-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
I can take this

marioprats, 2023-05-04:
All good.

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

clalancette, 2023-05-04:
Thank you for testing!