osrf/ros2_test_cases #1034: Writing A Tf2 Broadcaster Py

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Tf2 Broadcaster Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Tf2 Broadcaster Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Broadcaster-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
I can take this

marioprats, 2023-05-04:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [ ] The documentation didn't have any obvious errors.

marioprats, 2023-05-04:
There is an error in that tutorial, likely coming from previous versions.

The instructions to install to launch file should be:

```
data_files=[
    ...
    (os.path.join('share', package_name, 'launch'), glob(os.path.join('launch', '*launch.py'))),
],
```

Or either change the launch file name to be `turtle_tf2_demo.launch.py` instead of `turtle_tf2_demo_launch.py`.

As it is now, the glob pattern to install `*.launch.py` doesn't match the launch file name, which ends with `_launch.py`.

Let me know if you want me to update the tutorial.

clalancette, 2023-05-04:
Since we've now fixed the documentation issue here, I'm going to go ahead and close this out.  Thanks for the testing!

closed 2023-05-04 by clalancette