osrf/ros2_test_cases #1036: Writing A Tf2 Listener Py

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Tf2 Listener Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Tf2 Listener Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
I'll take this too given I'm doing all the other tf tutorials

marioprats, 2023-05-04:
Everything was fine.
I experienced [ros2/rclpy#1081](https://github.com/ros2/rclpy/issues/1081), but other than that all good.

clalancette, 2023-05-04:
> I experienced [ros2/rclpy#1081](https://github.com/ros2/rclpy/issues/1081), but other than that all good.

Yeah, we are seeing that everywhere.  I'll still call this good, thanks for testing!

closed 2023-05-04 by clalancette