osrf/ros2_test_cases #1037: Writing A Tf2 Static Broadcaster Cpp

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Tf2 Static Broadcaster Cpp' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Tf2 Static Broadcaster Cpp page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-03:
All good!

- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

Yadunund, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by Yadunund