osrf/ros2_test_cases #1038: Writing A Tf2 Static Broadcaster Py

Yadunund, 2023-05-01:
Check the documentation for the 'Writing A Tf2 Static Broadcaster Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Writing A Tf2 Static Broadcaster Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-03:
I can take this.

Tested on jammy and amd64. Worked fine, just two minor issues:

1) The build triggers a python warning:

```
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
```

2) Ctrl-C on the `static_turtle_tf2_broadcaster` doesn't shutdown cleanly:

```
ros2 run learning_tf2_py static_turtle_tf2_broadcaster mystaticturtle 0 0 1 0 0 0
(Ctrl-C)
Traceback (most recent call last):
  File "/ws/install/learning_tf2_py/lib/learning_tf2_py/static_turtle_tf2_broadcaster", line 33, in <module>
    sys.exit(load_entry_point('learning-tf2-py==0.0.0', 'console_scripts', 'static_turtle_tf2_broadcaster')())
  File "/ws/install/learning_tf2_py/lib/python3.10/site-packages/learning_tf2_py/static_turtle_tf2_broadcaster.py", line 115, in main
    rclpy.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:313
[ros2run]: Process exited with failure 1

```

marioprats, 2023-05-03:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

Yadunund, 2023-05-04:
Thanks for testing this out! The rcl_shutdown error is a known issue https://github.com/ros2/rclpy/issues/1081 so I'll go ahead and consider this done

closed 2023-05-04 by Yadunund