osrf/ros2_test_cases #934: Access Controls

Yadunund, 2023-05-01:
Check the documentation for the 'Access Controls' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Access Controls page](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Access-Controls.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-09:
For this one, I am unsure where to find the file: ca.cert.pem. After following the steps at the start of [Setting up security](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Introducing-ros2-security.html#setting-up-security), I find that I don't have directories at `~/sros2_demo/demo_keys/public` and `~/sros2_demo/demo_keys/private`.

For Access Controls specifically (and not Security in general), I didn't have the following files after following the listed steps: `permissions.p7s` and `permissions.xml`.

Yadunund, 2023-05-18:
The tutorial page instructions needed to be updated so I opened https://github.com/ros2/ros2_documentation/pull/3634

The keystore folder is at `~/sors2_demo/demo_keystore` and not `~/sros2_demo/demo_keys`. I managed to test this out and it works so I'll close it. 

```bash
yadunund@ubuntu-22-04:~/sros2_demo/demo_keystore/enclaves/talker_listener/talker$ ls
cert.pem
governance.p7s
identity_ca.cert.pem
key.pem
permissions.p7s
permissions.xml
permissions_ca.cert.pem
```

closed 2023-05-18 by Yadunund