osrf/ros2_test_cases #936: Adding A Frame Py

Yadunund, 2023-05-01:
Check the documentation for the 'Adding A Frame Py' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Adding A Frame Py page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Adding-A-Frame-Py.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-08:
In a debian jammy installation with fastdds.

I have been doing the full TF2 tutorial, including the "Adding a frame (Python)" page.

I was able to follow the documentation, which seemed clear to me. I was able to execute the code and do the tutorial. I have not seen any obvious error in wording/code or hyperlink.

[image]

clalancette, 2023-05-08:
Thank you for testing!

closed 2023-05-08 by clalancette