osrf/ros2_test_cases #937: Adding Physical and Collision Properties to a URDF Model

Yadunund, 2023-05-01:
Check the documentation for the 'Adding Physical and Collision Properties to a URDF Model' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Adding Physical and Collision Properties to a URDF Model page](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Adding-Physical-and-Collision-Properties-to-a-URDF-Model.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Went through this tutorial as well.
All good, this one is just instructional, no code involved.

clalancette, 2023-05-04:
Thanks for testing!

closed 2023-05-04 by clalancette