osrf/ros2_test_cases #938: Advanced

Yadunund, 2023-05-01:
Check the documentation for the 'Advanced' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Advanced page](https://docs.ros.org/en/iron/Tutorials/Advanced.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-10:
This documentation page is correct, all listed items correctly point to matching pages, no issue in hyperlinks.

Yadunund, 2023-05-10:
Awesome, thanks for verifying!

closed 2023-05-10 by Yadunund