osrf/ros2_test_cases #940: Ament CMake Documentation

Yadunund, 2023-05-01:
Check the documentation for the 'Ament CMake Documentation' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Ament CMake Documentation page](https://docs.ros.org/en/iron/Tutorials/Ament-CMake-Documentation.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-16:
Was able to follow. No obvious errors.

Yadunund, 2023-05-16:
Awesome, thanks for testing!

closed 2023-05-16 by Yadunund