osrf/ros2_test_cases #941: Beginner CLI Tools

Yadunund, 2023-05-01:
Check the documentation for the 'Beginner CLI Tools' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Beginner CLI Tools page](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

Kartik9250, 2023-05-05:
Hey, I would like to take on this ticket

Kartik9250, 2023-05-05:
I went through the documentation and the following were my key findings:

1. [Using turtlesim,ros2, and rqt](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Introducing-Turtlesim/Introducing-Turtlesim.html#using-turtlesim-ros2-and-rqt):
- While closing rqt works but the following error pops out:
`^C[INFO] [1683275454.064393738] [rclcpp]: signal_handler(signum=2)
rclpy.shutdown() was called before QThread.quit()
PluginHandler.shutdown_plugin() plugin "rqt_service_caller/ServiceCaller#1" raised an exception:
Traceback (most recent call last):
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 136, in shutdown_plugin
    self._shutdown_plugin()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler_direct.py", line 92, in _shutdown_plugin
    self.emit_shutdown_plugin_completed()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_handler.py", line 150, in emit_shutdown_plugin_completed
    callback(self._instance_id)
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 480, in _close_application_shutdown_callback
    self._close_application_signal()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/plugin_manager.py", line 483, in _close_application_signal
    self._plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/qt_gui/composite_plugin_provider.py", line 87, in shutdown
    plugin_provider.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 62, in shutdown
    self._destroy_node()
  File "/opt/ros/iron/lib/python3.10/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 89, in _destroy_node
    rclpy.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/iron/lib/python3.10/site-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:313`

- Closing it through gui works but gives this error:
`Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
         at line 127 in ./src/class_loader.cpp`
similar issue was mention in #1048 

2. [Understanding nodes](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Nodes/Understanding-ROS2-Nodes.html#understanding-nodes):
- In [2.1 remapping](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Nodes/Understanding-ROS2-Nodes.html#remapping), the 'cmd_vel' should be highlighted for better readability.
- rqt error as mentioned above

3. [Understanging topics](https://docs.ros.org/en/iron/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Topics/Understanding-ROS2-Topics.html#understanding-topics):
- Difference in the rqt graph as compared to the wiki. Attached below are the graphs that I generated
    - ['2 rqt graph' graph](https://ibb.co/HpvDqkx)
    - ['3 ros2 topic list' graph](https://ibb.co/1sTv3dK)
    - ['4 ros2 topic echo' graph](https://ibb.co/Fbv48BV)
    - ['5 ros2 topic info' graph](https://ibb.co/0jvNCSk)
    - '7 ros2 topic pub' [graph1](https://ibb.co/8jPSpXC), [graph2](https://ibb.co/tHjB1T7)
    
## Checks
- [x] I was able to follow the documentation.

- [x] The documentation seemed clear to me.

- [x] The documentation didn't have any obvious errors.

Yadunund, 2023-05-09:
Thanks for testing! Suggest opening PRs to ros2/ros2_documentation to improve on 2) & 3).
1) & 2) are being tracked. I'll close this ticket