osrf/ros2_test_cases #942: Beginner Client Libraries

Yadunund, 2023-05-01:
Check the documentation for the 'Beginner Client Libraries' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Beginner Client Libraries page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-10:
This page is correct, all listed items  match their names, no issue found in hyperlinks.

clalancette, 2023-05-10:
Thank you for testing!

closed 2023-05-10 by clalancette