osrf/ros2_test_cases #943: Building a Movable Robot Model with URDF

Yadunund, 2023-05-01:
Check the documentation for the 'Building a Movable Robot Model with URDF' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Building a Movable Robot Model with URDF page](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Building-a-Movable-Robot-Model-with-URDF.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Works but same comment as in https://github.com/osrf/ros2_test_cases/issues/944 regarding the TF display being enabled by default, whereas the image in the tutorial has it disabled.

Yadunund, 2023-05-05:
Thanks for testing. Left a comment on the other ticket regarding the TF.