osrf/ros2_test_cases #944: Building a Visual Robot Model with URDF from Scratch

Yadunund, 2023-05-01:
Check the documentation for the 'Building a Visual Robot Model with URDF from Scratch' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Building a Visual Robot Model with URDF from Scratch page](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
Starting on the URDF tutorials.

marioprats, 2023-05-04:
All good.
One observation is that all the images in the tutorial have the TF display disabled, whereas when you run this the TF display is enabled by default, per config. Probably not a big deal but leaving a note here.

[image]

Yadunund, 2023-05-05:
Looking through the images in the tutorial docs, the TF is not enabled in any of the sections except the last one https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html. In this section, the image shows that the TF display is checked. It's probably worth adding a note to the [View the results](https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html#view-the-results) section. Do you mind opening a PR [here](https://github.com/ros2/ros2_documentation/) to add a note about enabling the TF display? 

I'll consider this testing done!

closed 2023-05-05 by Yadunund

marioprats, 2023-05-05:
Hey Yadu, the TF display is enabled by default, so that tutorial is fine (https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html).

The confusion can come in this other tutorial, in the last image:
https://docs.ros.org/en/iron/Tutorials/Intermediate/URDF/Building-a-Visual-Robot-Model-with-URDF-from-Scratch.html

The tutorial image has TF disabled, whereas if your run this on your computer, TF will be enabled by default. So someone may get confused about the visualization being different. Again I don't think this is a big deal, so we can probably leave as-is. Another option would be to update the tutorial image to have TF enabled, or to add a comment about enabling/disabling TF.