osrf/ros2_test_cases #945: Building ROS2 Package with eclipse 2021 06

Yadunund, 2023-05-01:
Check the documentation for the 'Building ROS2 Package with eclipse 2021 06' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Building ROS2 Package with eclipse 2021 06 page](https://docs.ros.org/en/iron/Tutorials/Miscellaneous/Building-ROS2-Package-with-eclipse-2021-06.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-16:
I'll take this one.

iuhilnehc-ynos, 2023-05-16:
* [ ]  **I was able to follow the documentation.**
* [ ]  **The documentation seemed clear to me.**
* [ ]  **The documentation didn't have any obvious errors.**

Issues:

- `ROS 2 includes` could not be resolved with `/opt/ros/iron/include`
  I need to use paths of separate packages. e.g. `/opt/ros/iron/include/rclcpp`, `/opt/ros/iron/include/std_msgs`.
- Build successfully, but
  Eclipse uses `colcon` to build `/home/ros2/dev_ws` rather than the current package(`/home/ros2/eclipse-workspace/my_package`) in the workspace, which means if I update the source code with Eclipse and then build again, it can't be effective.
  The reason is that the package is imported with a copy operation, not with a link.

<details><summary> log </summary>
<p>

- ROS 2 includes got resolved (NG)
[image]
[image]
The header files are installed with [DESTINATION include/${PROJECT_NAME}](https://github.com/ros2/rclcpp/blob/a7048e115d9316b54161180fa3bddab29bce7cc3/rclcpp/CMakeLists.txt#L265)
[image]

- Import files without using a link, but to build for another working directory
[image]
[image]

</p>
</details>

iuhilnehc-ynos, 2023-05-16:
@karmenlu, I just shared my test. 

I appreciate that if you could review and confirm the document as well.

Yadunund, 2023-05-18:
@iuhilnehc-ynos thanks for testing this out! Nice catch of the discrepancy between the name of the workspace in the first screenshot and the working directory set later on. And also the point about having to add include paths for each package separately.

Do you mind opening a PR to update the page on ros2_documentation?

iuhilnehc-ynos, 2023-05-19:
> Do you mind opening a PR to update the page on ros2_documentation?

No, I'd love to.

Yadunund, 2023-05-19:
Awesome, will close this out.

closed 2023-05-19 by Yadunund