osrf/ros2_test_cases #946: CLI

Yadunund, 2023-05-01:
Check the documentation for the 'CLI' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [CLI page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Testing/CLI.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-08:
On a debian jammy installation with fastDDS.

I followed the Test-CLI documentation, as part of the Running Tests tutorial.

I was able to follow the documentation, it was cleared to me and I did not see any obvious error.

clalancette, 2023-05-08:
Thank you for testing!