osrf/ros2_test_cases #947: Colcon Tutorial

Yadunund, 2023-05-01:
Check the documentation for the 'Colcon Tutorial' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Colcon Tutorial page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

cdhabecker, 2023-05-04:
I'll take this one.

cdhabecker, 2023-05-04:
**Observations**:
1. This tutorial appears to be the major colcon doc in docs.ros.org. (I searched "colcon" in the "Search docs" field, and the hits are mostly colcon command lines.) In fact, https://docs.ros.org/en/iron/Installation/Alternatives/Ubuntu-Development-Setup.html refers to this tutorial for "More info on working with a ROS workspace...".

**Problems**
1. "This is a brief tutorial on how to create and build a ROS 2 workspace with colcon. It is a practical tutorial and not designed to replace the core documentation." -> "core documentation" should be a link to https://colcon.readthedocs.io/.
2. "Background" section -> delete. https://colcon.readthedocs.io/ explains the relationship to catkin and gives the github link. As for the [design document](https://design.ros2.org/articles/build_tool.html), not even https://colcon.readthedocs.io/ mentions it.
3. "Basics" section: too long, too much passive voice. Replace the entire section with: "A ROS workspace is a directory with a particular subdirectory structure. You place the source code packages in the src/ directory and colcon builds the source code into the build/, install/, and log/ directories.

cdhabecker, 2023-05-04:
**Checks**
 Other than the problems in the preceding comment:
- [x] I was able to follow the documentation.
- [x]  The documentation seemed clear to me.
- [x]  The documentation didn't have any obvious errors.

**Tests passed**
The build and test commands printed a bunch of known warnings as per https://github.com/colcon/colcon-core/issues/454#issuecomment-1319121948. Nevertheless, the tests passed.

**The demo worked.**

**Shell output**
https://gist.github.com/cdhabecker/942489fb7469bab375845de20591e79f

Yadunund, 2023-05-04:
Thanks for testing this! Appreciate the valuable comments on improving documentation. Will continue tracking this and other suggestions to docs here https://github.com/osrf/ros2_test_cases/discussions/1047

I'll close this ticket as its considered completed

closed 2023-05-04 by Yadunund