osrf/ros2_test_cases #948: Composition

Yadunund, 2023-05-01:
Check the documentation for the 'Composition' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Composition page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Composition.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-08:
In a debian jammy installation with fastDDS.

I have followed the documentation page for "Composing multiple nodes in a single process"

I was able to follow the documentation and run the different code sections. The documentation was clear to me, and I have not seen any obvious errors.

I would just propose a slight improvment ( not-compulsory at all) to make all "Run the demos" sections coherent, it is to add a link to the composition_demo_launch.py source code in the "Composition using launch actions" section, with something like:  [source code](https://github.com/ros2/demos/blob/iron/composition/launch/composition_demo_launch.py)
All other "Run the demos" have pointer to the source code, which I feel useful.

Screenshots of my session:
[image]
[image]
[image]
[image]

clalancette, 2023-05-08:
> I would just propose a slight improvment ( not-compulsory at all) to make all "Run the demos" sections coherent, it is to add a link to the composition_demo_launch.py source code in the "Composition [...]

Good catch!  I agree that is useful.  I've opened up https://github.com/ros2/ros2_documentation/pull/3553 to add this in.

With that said, the rest of this looks good, so I'll close this out.  Thank you for testing!

closed 2023-05-08 by clalancette