osrf/ros2_test_cases #951: Cpp

Yadunund, 2023-05-01:
Check the documentation for the 'Cpp' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Cpp page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Writing-an-Action-Server-Client/Cpp.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

BorisBoutillier, 2023-05-08:
On a debian jammy installation with fastDDS.

I have followed the documentation for Writing an action server/client in CPP.

I was able to follow the documentation. It was clear to me. It did not have obvious errors in wording/code or link pages.
I was able to create compile and run the server and client.

Screenshot:
[image]

Yadunund, 2023-05-08:
Awesome, thank you!

closed 2023-05-08 by Yadunund