osrf/ros2_test_cases #952: Creating A Workspace

Yadunund, 2023-05-01:
Check the documentation for the 'Creating A Workspace' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Creating A Workspace page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

sea-bass, 2023-05-03:
I ran through this not with the demos shown above, but instead building [MoveIt 2 Tutorials from source](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) in a new workspace, including `rosdep` install etc. Worked perfectly.

clalancette, 2023-05-03:
Thanks for testing!

cdhabecker, 2023-05-04:
In section "4 Resolve dependencies", the phrase "you will need to use the rosdep command from their installation instructions" raises questions. What does it even mean? The paragraph provides this explanation:

>  "Here are the [from-source rosdep section](https://docs.ros.org/en/iron/Installation/Alternatives/Ubuntu-Development-Setup.html#linux-development-setup-install-dependencies-using-rosdep) and the [“fat” archive rosdep section](https://docs.ros.org/en/iron/Installation/Alternatives/Ubuntu-Install-Binary.html#linux-install-binary-install-missing-dependencies)."

but each of those instructions installs rosdep as a Debian package (either directly or via ros-dev-tools), which means that the default path of the user's default environment simply includes rosdep. Therefore, those instructions don't explain how to use rosdep. On the other hand, those instructions do explain how to use a particular installation of ROS (e.g., ". ~/ros2_iron/install/local_setup.bash"). But rosdep isn't part of a ROS installation, it's a tool that works on any ROS installation.

**Conclusion**: delete this paragraph.

cdhabecker, 2023-05-08:
**The given sample fails to clone because there is no iron-devel branch of the repo ros_tutorials.**

@Yadunund, @tylerjw

```
$ git clone https://github.com/ros/ros_tutorials.git -b iron-devel
Cloning into 'ros_tutorials'...
fatal: Remote branch iron-devel not found in upstream origin
```

cdhabecker, 2023-05-08:
**The given build statement prints a warning that the tutorial does not mention.**

In section 5 "Build the workspace with colcon", the tutorial says that the output should be:

```
Starting >>> turtlesim
Finished <<< turtlesim [5.49s]

Summary: 1 package finished [5.58s]
```
but the actual output is:
```
habecker@2023-05-07 18:29:16:~/ros2_ws$ colcon build
[0.215s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'turtlesim' is in: /opt/ros/iron
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding turtlesim

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> turtlesim
Finished <<< turtlesim [12.1s]

Summary: 1 package finished [12.2s]
habecker@2023-05-07 18:29:41:~/ros2_ws$ 
```
Jammy, amd64

Yadunund, 2023-05-08:
> **The given sample fails to clone because there is no iron-devel branch of the repo ros_tutorials.**
> 
> @Yadunund, @tylerjw
> 
> ```
> $ git clone https://github.com/ros/ros_tutorials.git -b iron-devel
> Cloning into 'ros_tutorials'...
> fatal: Remote branch iron-devel not found in upstream origin
> ```

Thanks for highlighting this. I've just pushed an `iron-devel` branch so could you try again?

Yadunund, 2023-05-08:
> **The given build statement prints a warning that the tutorial does not mention.**
>
> In section 5 "Build the workspace with colcon", the tutorial says that the output should be:
>
> ```
> Starting >>> [...]

This is similar to https://github.com/ros2/ros2_documentation/issues/3531

cdhabecker, 2023-05-08:
> Thanks for highlighting this. I've just pushed an `iron-devel` branch so could you try again?

**I tried again with "iron-devel" rather than "iron". "iron-devel" works now. (colcon build output still shows the warning.)**

cdhabecker, 2023-05-08:
**Remainder of tutorial worked for me.**

jammy, amd64, debian

Yadunund, 2023-05-08:
Thanks for testing. I'll close this ticket and follow up on the colcon overlap issue separately.

closed 2023-05-08 by Yadunund