osrf/ros2_test_cases #953: Creating an Action

Yadunund, 2023-05-01:
Check the documentation for the 'Creating an Action' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Creating an Action page](https://docs.ros.org/en/iron/Tutorials/Actions/Creating-an-Action.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-04:
I'll take this one too, since the other action server tutorials depend on this.

marioprats, 2023-05-04:
All good.
I got a warning when building because `action_tutorials_interfaces` is apparently included in the desktop debian packages, so I had that package already under `/opt/ros/iron`.

```
[0.213s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'action_tutorials_interfaces' is in: /opt/ros/iron
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding action_tutorials_interfaces

This may be promoted to an error in a future release of colcon-override-check.
```

Yadunund, 2023-05-04:
This is a known issue with colcon where the package with the same name is sourced in the underlay.

Do you mind testing this again after doing either one of these:
1. `sudo apt remove ros-iron-action-tutorials-interfaces` and rebuild OR
2. Rename the package you created to something else (might be easier)

@clalancette we should probably update the doc page to guide users to give a different name to the package so that the underlay issue is not encountered?

Yadunund, 2023-05-05:
Thanks for testing! I've documented the colcon underlay issue for resolution in the near future https://github.com/ros2/ros2_documentation/issues/3531. I'll consider this test done

closed 2023-05-05 by Yadunund