osrf/ros2_test_cases #956: Custom ROS2 Interfaces

Yadunund, 2023-05-01:
Check the documentation for the 'Custom ROS2 Interfaces' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Custom ROS2 Interfaces page](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Custom-ROS2-Interfaces.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jjd9, 2023-05-05:
I would like to do this one.

> I was able to follow the documentation.

Yes, it guided me through the exercise smoothly.

> The documentation seemed clear to me.

Mostly. One exception that might trip some people up is that in section 4 `package.xml`, I think it would be a good idea to show a completed/minimal package.xml. Users might not know which parts of the autogenerated package.xml they need to keep and which ones aren't needed when only building ROS2 interfaces.

e.g.

```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>tutorial_interfaces</name>
  <version>0.0.0</version>
  <description>TODO: Package description</description>
  <maintainer email="email@gmail.com">Maintainer1</maintainer>
  <license>TODO: License declaration</license>

  <depend>geometry_msgs</depend>
  <buildtool_depend>rosidl_default_generators</buildtool_depend>
  <exec_depend>rosidl_default_runtime</exec_depend>
  <member_of_group>rosidl_interface_packages</member_of_group>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

> The documentation didn't have any obvious errors.

Yes, I did not find any obvious errors. I tested the C++ and Python instructions (on a amd64 machine + debian ros2 iron install + fastdds rmw + ubuntu jammy) and they worked as described in the tutorial.

However, there is still this deprecation warning when building the package using the Python instructions. But that is a known ROS2 issue afaik and isnt relevant to this tutorial (but maybe worth mentioning in the python pub/sub or service/client tutorials?)
```
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
```

Would it be helpful for me to paste the terminal output of all the examples from the tutorial (the c++ pub/sub, C++ service/client, python pub/sub, python service client) ?

Yadunund, 2023-05-05:
Thanks for testing! I'd encourage you to open a PR to ros2/ros2_documentation to also include the full `package.xml`.

The setuptools warning is expected so I'll go ahead an close this ticket.

closed 2023-05-05 by Yadunund

jjd9, 2023-05-05:
opened PR :+1: 
https://github.com/ros2/ros2_documentation/pull/3534