osrf/ros2_test_cases #957: Custom ROS2 Interfaces.rst

Yadunund, 2023-05-01:
Check the documentation for the 'Custom ROS2 Interfaces.rst' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Custom ROS2 Interfaces.rst page](https://docs.ros.org/en/iron/Tutorials/Custom-ROS2-Interfaces.rst.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jacobperron, 2023-05-08:
Duplicate of https://github.com/osrf/ros2_test_cases/issues/956

Yadunund, 2023-05-11:
Thanks for catching that. Will take note for the next iteration.

closed 2023-05-11 by Yadunund