osrf/ros2_test_cases #958: Debugging Tf2 Problems

Yadunund, 2023-05-01:
Check the documentation for the 'Debugging Tf2 Problems' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Debugging Tf2 Problems page](https://docs.ros.org/en/iron/Tutorials/Intermediate/Tf2/Debugging-Tf2-Problems.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

marioprats, 2023-05-08:
This tutorial is based on code from another tutorial.
There were some mismatches in the instructions in this tutorial, probably due to changes in the other tutorial code.
Here's a PR to update this tutorial as needed: https://github.com/ros2/ros2_documentation/pull/3557

Other than that, all good.

clalancette, 2023-05-08:
Since we have that separate PR open, I'm going to close this one out.

Thank you for testing!

closed 2023-05-08 by clalancette