osrf/ros2_test_cases #959: Demos

Yadunund, 2023-05-01:
Check the documentation for the 'Demos' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Demos page](https://docs.ros.org/en/iron/Tutorials/Demos.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

christophebedard, 2023-05-16:
There isn't a lot to validate here, but:

* [x] I was able to follow the documentation.
* [x] The documentation seemed clear to me.
* [ ] The documentation didn't have any obvious errors.
   * The "Motion planning for a MoveIt 2 arm" link is being redirected from http://moveit2_tutorials.picknik.ai/ to https://moveit.picknik.ai/humble/index.html, so this could probably be updated too
   * The "Using Turtlebot 2" link could probably be removed, since it's old and the repository has been archived

I'll open a PR.

clalancette, 2023-05-17:
Thanks for testing!

Since you've already opened PRs for the mentioned issues, I'll close this out.

closed 2023-05-17 by clalancette