osrf/ros2_test_cases #960: Deploying ROS 2 on IBM Cloud

Yadunund, 2023-05-01:
Check the documentation for the 'Deploying ROS 2 on IBM Cloud' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Deploying ROS 2 on IBM Cloud page](https://docs.ros.org/en/iron/Tutorials/Miscellaneous/Deploying-ROS-2-on-IBM-Cloud.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

pradyum, 2023-05-18:
Hi, there is no iron docker image released yet. Should I this documentation test with foxy, rolling or humble image?

Yadunund, 2023-05-18:
Please use rolling!

clalancette, 2023-06-23:
This one didn't get testing, but I think we are going to skip it next year anyway so closing this out.

closed 2023-06-23 by clalancette