osrf/ros2_test_cases #961: Deployment Guidelines

Yadunund, 2023-05-01:
Check the documentation for the 'Deployment Guidelines' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Deployment Guidelines page](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Deployment-Guidelines.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

jmackay2, 2023-05-10:
- [x] I was able to follow the documentation.
- [x] The documentation seemed clear to me.
- [x] The documentation didn't have any obvious errors.

Outside of the PR made to update the Dockerfile to default to use `iron` instead of `humble`, I didn't have any issues with the documentation.

Yadunund, 2023-05-10:
Thanks for testing!

closed 2023-05-10 by Yadunund