osrf/ros2_test_cases #963: Discovery Server

Yadunund, 2023-05-01:
Check the documentation for the 'Discovery Server' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Discovery Server page](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

OmkarKabadagi5823, 2023-05-07:
I will take this up.

OmkarKabadagi5823, 2023-05-08:
All the examples apart from [compare-fast-dds-discovery-server-with-simple-discovery-protocol](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#compare-fast-dds-discovery-server-with-simple-discovery-protocol) worked as expected.

When running the `generate_discovery_packages.bash` for simple discovery and discovery server setups, contrary to what is expected, the number of messages in the simple discovery protocol was a lot fewer than in the discover server protocol.

```
source to file:  /opt/ros/iron/setup.bash
Run in Simple Discovery mode
Spawn talker
Spawn first listener 0
Traffic capture can be found in: simple.pcapng
source to file:  /opt/ros/iron/setup.bash
Run in Discovery Server mode
Spawn talker
Spawn first listener 0
Traffic capture can be found in: discovery_server.pcapng
Processing simple.pcapng
Processing discovery_server.pcapng
  discovery_protocol  discovery_packets
0                     Simple               6432
1   Discovery Server              84004
```

[image]

OmkarKabadagi5823, 2023-05-08:
All the examples apart from [compare-fast-dds-discovery-server-with-simple-discovery-protocol](https://docs.ros.org/en/iron/Tutorials/Advanced/Discovery-Server/Discovery-Server.html#compare-fast-dds-discovery-server-with-simple-discovery-protocol) worked as expected.

When running the `generate_discovery_packages.bash` for simple discovery and discovery server setups, contrary to what is expected, the number of messages in the simple discovery protocol was a lot fewer than in the discover server protocol.

```
source to file:  /opt/ros/iron/setup.bash
Run in Simple Discovery mode
Spawn talker
Spawn first listener 0
[INFO] [1683530557.887479581] [simple_talker]: Publishing: 'Hello World: 1'
[INFO] [1683530557.888996092] [listener_0]: I heard: [Hello World: 1]
[INFO] [1683530558.887367700] [simple_talker]: Publishing: 'Hello World: 2'
[INFO] [1683530558.907329075] [listener_0]: I heard: [Hello World: 2]
[INFO] [1683530559.887463503] [simple_talker]: Publishing: 'Hello World: 3'
[INFO] [1683530559.936608561] [listener_0]: I heard: [Hello World: 3]
[INFO] [1683530560.887373423] [simple_talker]: Publishing: 'Hello World: 4'
[INFO] [1683530560.889425443] [listener_0]: I heard: [Hello World: 4]
[INFO] [1683530561.887358056] [simple_talker]: Publishing: 'Hello World: 5'
[INFO] [1683530561.887966097] [listener_0]: I heard: [Hello World: 5]
[INFO] [1683530562.887553298] [simple_talker]: Publishing: 'Hello World: 6'
[INFO] [1683530562.888521941] [listener_0]: I heard: [Hello World: 6]
[INFO] [1683530563.887552417] [simple_talker]: Publishing: 'Hello World: 7'
[INFO] [1683530563.888270475] [listener_0]: I heard: [Hello World: 7]
[INFO] [1683530564.887550678] [simple_talker]: Publishing: 'Hello World: 8'
[INFO] [1683530564.888544108] [listener_0]: I heard: [Hello World: 8]
[INFO] [1683530565.887541840] [simple_talker]: Publishing: 'Hello World: 9'
[INFO] [1683530565.888746222] [listener_0]: I heard: [Hello World: 9]
[INFO] [1683530566.887490788] [simple_talker]: Publishing: 'Hello World: 10'
[INFO] [1683530566.888559677] [listener_0]: I heard: [Hello World: 10]
[INFO] [1683530567.887564218] [simple_talker]: Publishing: 'Hello World: 11'
[INFO] [1683530567.888504936] [listener_0]: I heard: [Hello World: 11]
[INFO] [1683530568.887541064] [simple_talker]: Publishing: 'Hello World: 12'
[INFO] [1683530568.888928658] [listener_0]: I heard: [Hello World: 12]
[INFO] [1683530569.887458950] [simple_talker]: Publishing: 'Hello World: 13'
[INFO] [1683530569.888838896] [listener_0]: I heard: [Hello World: 13]
[INFO] [1683530570.887537562] [simple_talker]: Publishing: 'Hello World: 14'
[INFO] [1683530570.888959748] [listener_0]: I heard: [Hello World: 14]
[INFO] [1683530571.784201083] [rclcpp]: signal_handler(signum=2)
[INFO] [1683530571.812831958] [rclcpp]: signal_handler(signum=2)
Traffic capture can be found in: simple.pcapng
source to file:  /opt/ros/iron/setup.bash
Run in Discovery Server mode
Spawn talker
Spawn first listener 0
[INFO] [1683530583.860857217] [server_talker]: Publishing: 'Hello World: 1'
[INFO] [1683530583.861810757] [listener_0]: I heard: [Hello World: 1]
[INFO] [1683530584.860862844] [server_talker]: Publishing: 'Hello World: 2'
[INFO] [1683530584.862055085] [listener_0]: I heard: [Hello World: 2]
[INFO] [1683530585.860846332] [server_talker]: Publishing: 'Hello World: 3'
[INFO] [1683530585.861053015] [listener_0]: I heard: [Hello World: 3]
[INFO] [1683530586.860965449] [server_talker]: Publishing: 'Hello World: 4'
[INFO] [1683530586.878563482] [listener_0]: I heard: [Hello World: 4]
[INFO] [1683530587.861066187] [server_talker]: Publishing: 'Hello World: 5'
[INFO] [1683530587.861755748] [listener_0]: I heard: [Hello World: 5]
[INFO] [1683530588.861085689] [server_talker]: Publishing: 'Hello World: 6'
[INFO] [1683530588.863457785] [listener_0]: I heard: [Hello World: 6]
[INFO] [1683530589.861057979] [server_talker]: Publishing: 'Hello World: 7'
[INFO] [1683530589.861694315] [listener_0]: I heard: [Hello World: 7]
[INFO] [1683530590.861067016] [server_talker]: Publishing: 'Hello World: 8'
[INFO] [1683530590.862154739] [listener_0]: I heard: [Hello World: 8]
[INFO] [1683530591.860928066] [server_talker]: Publishing: 'Hello World: 9'
[INFO] [1683530591.862001463] [listener_0]: I heard: [Hello World: 9]
[INFO] [1683530592.861042556] [server_talker]: Publishing: 'Hello World: 10'
[INFO] [1683530592.861623544] [listener_0]: I heard: [Hello World: 10]
[INFO] [1683530593.861076541] [server_talker]: Publishing: 'Hello World: 11'
[INFO] [1683530593.862036040] [listener_0]: I heard: [Hello World: 11]
[INFO] [1683530594.861012243] [server_talker]: Publishing: 'Hello World: 12'
[INFO] [1683530594.861511642] [listener_0]: I heard: [Hello World: 12]
[INFO] [1683530595.860878929] [server_talker]: Publishing: 'Hello World: 13'
[INFO] [1683530595.862667108] [listener_0]: I heard: [Hello World: 13]
[INFO] [1683530596.491011309] [rclcpp]: signal_handler(signum=2)
[INFO] [1683530596.512395692] [rclcpp]: signal_handler(signum=2)
Traffic capture can be found in: discovery_server.pcapng
Processing simple.pcapng
Processing discovery_server.pcapng
  discovery_protocol  discovery_packets
0             Simple               6432
1   Discovery Server              84004
```

[image]

Yadunund, 2023-05-11:
Thanks a lot for trying this out! Do you mind opening a ticket in the FastDDS repo with your test results? The developers may have better insights on why the results here vary from the documentation. But since the instructions on the page all work, I'll close this ticket,.

closed 2023-05-11 by Yadunund