osrf/ros2_test_cases #965: Examine Traffic

Yadunund, 2023-05-01:
Check the documentation for the 'Examine Traffic' page

## Setup
- DDS vendor: FastDDS
- BuildType: Debian
- Os: Ubuntu Jammy
- Chip: Amd64

## Links
- [Examine Traffic page](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Examine-Traffic.html)

## Checks
- [ ] **I was able to follow the documentation.**
- [ ] **The documentation seemed clear to me.**
- [ ] **The documentation didn't have any obvious errors.**

---

*You can find the code used to generate this test case [here](https://github.com/audrow/yatm)*

karmenlu, 2023-05-09:
Suggested revision for 'Install tcp' section:
```
sudo apt update
sudo apt upgrade
sudo apt install tcpdump
```

- Able to display unencrypted discovery packets.
- Unable to display unencrypted data packets after running `sudo tcpdump -i any -X udp portrange 7401-7500`
- Able to display encrypted discovery packets.
- Unable to display encrypted data packets after running `sudo tcpdump -i any -X udp portrange 7401-7500`

Nothing shows up for both unencrypted and encrypted data packet sections:
[image]

This page should also indicate that following the steps on [Setting up security ](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Introducing-ros2-security.html) are a prerequisite for the [Enable encryption](https://docs.ros.org/en/iron/Tutorials/Advanced/Security/Examine-Traffic.html#enable-encryption) section.

Yadunund, 2023-05-16:
Thanks for trying this out. I agree the test case should better inform users to follow the previous sections of the Security tutorial and have taken note to improve this for next time https://github.com/audrow/yatm/issues/366

Noted on not being able to display encrypted or unencrypted data packages. Could I check the version of `openssl` python3-cryptography` you're using?

fujitatomoya, 2023-05-19:
> Noted on not being able to display encrypted or unencrypted data packages.

this can be reproducible for fast-dds, using cyclone we can see the data payload via UDP with `tcpdump -i any -X udp portrange 7401-7500`

we can see fast-dds listens on the following ports, as expected.

```bash
root@tomoyafujita:~# netstat -tulpn | grep -e talker -e listener
udp        0      0 0.0.0.0:7400            0.0.0.0:*                           3304/listener
udp        0      0 0.0.0.0:7400            0.0.0.0:*                           3289/talker
udp        0      0 0.0.0.0:7410            0.0.0.0:*                           3289/talker
udp        0      0 0.0.0.0:7411            0.0.0.0:*                           3289/talker
udp        0      0 0.0.0.0:7412            0.0.0.0:*                           3304/listener
udp        0      0 0.0.0.0:7413            0.0.0.0:*                           3304/listener
udp        0      0 0.0.0.0:57153           0.0.0.0:*                           3289/talker
udp        0      0 43.135.146.89:59268     0.0.0.0:*                           3289/talker
udp        0      0 10.0.0.137:60530        0.0.0.0:*                           3289/talker
udp        0      0 43.135.146.89:45435     0.0.0.0:*                           3304/listener
udp        0      0 10.0.0.137:47449        0.0.0.0:*                           3304/listener
udp        0      0 0.0.0.0:47705           0.0.0.0:*                           3304/listener
```

but never put the data on the network interface, i guess. this could be related to share memory transport or data sharing. (expecting that data sharing cannot be enabled since string is not PoD.)

@MiguelCompany any thoughts?

MiguelCompany, 2023-05-23:
@fujitatomoya 
> this could be related to share memory transport

Yes. Running on the same host will by default use the shared memory transport, so no traffic will be seen on udp ports `7401-7500`.

You would see the traffic if the talker and listener are run on different hosts.

fujitatomoya, 2023-05-24:
@MiguelCompany thanks for the confirmation.

@Yadunund i will try to come up with PR for this, probably adding some note for `rmw_fastrtps`? that makes sense to you?

fujitatomoya, 2023-06-05:
@Yadunund we can close this, addressed in https://github.com/ros2/ros2_documentation/pull/3683.

CC: @clalancette

Yadunund, 2023-06-06:
Awesome, thanks for taking care of that!

closed 2023-06-06 by Yadunund